asrob-uc3m/robotDevastation-robots #2: Upload image  of RD1 SD card.

jgvictores, 2015-01-30:
Can be uploaded as file in SF. Must set: default enable_pwmbot true, rc.local (before creating image).

David-Estevez, 2015-01-30:
@jgvictores What about making a 'distro' with everything ready / almost ready to just write the image in the sd card and play? Support to other platforms (Beaglebone, etc)?

jgvictores, 2015-01-30:
Yes, what I had in mind is getting what we already have (a raspian with launchRd1), set rc.local to launch launchRd1 (maybe through yarprun), and uploading that plug-and-play image (fsa  and/or raw dd).
It would be great to set up an apt-ppa, and port to many platforms. Let's keep this issue for keeping track of the rd1-raspian image, and open new issues for specific ports.

jgvictores, 2015-02-01:
Should check  tools that offer the possibility of a graphical front-end (Clonezilla, perhaps?).

David-Estevez, 2015-02-01:
A graphical front-end for uploading the image to the SD card? I used Clonezilla for backups in the past, and you still need to boot Clonezilla for that, so you have to write Clonezilla on another device.

Unetbootin worked for me when writing Live USBs from ISOs, but I guess it does not work the same way with Raspberry Pi images

jgvictores, 2015-02-08:
Two comments regarding deployment:
- I believe defaulting to using a yarprun daemon on the robots will prove useful. Yarprun is a server to launch and monitor programs remotely. This way, the robotDevastation PC can have all the setup options, and launch the corresponding robotDevastation-robots program with the correct arguments.
- We should consider Docker Hub Registry (https://registry.hub.docker.com/). Check the following currently ongiong thread: http://ros-users.122217.n3.nabble.com/Givving-ROS-an-Official-DockerHub-Image-td4021623.html

David-Estevez, 2015-02-08:
The daemon seems to be a good choice, this way the robot could be managed directly from the player's PC.

I didn't know about Docker, what a nice concept! We could definitely give in a try for shipping the game.

jgvictores, 2016-04-20:
The issue of the title has been closed a while ago (http://asrob.uc3m.es/index.php/C%C3%B3mo_montar_el_robot_RD2#Crear_SD_con_Raspbian_y_robotDevastation-robots.2Fraspi_software), so closing.

closed 2016-04-20 by jgvictores