asrob-uc3m/robotDevastation-robots #3: Have a usable and documented robot.

jgvictores, 2015-02-02:
Both rd1 and rd2 look like good candidates, but we need documentation on construction and installing software too for anyone to build it.

Rodriwp, 2015-02-02:
I think we must create documentation for rd2 ( with raspberry pi) as the standard robot for everyone and documentation for the "Ardubot module" that enables anyone with arduino and robotics skills to implement robot devastation in any project without linux skills or advance programming skills.

David-Estevez, 2015-02-02:
I would argue that controlling the board's GPIOs with python requires less skills than developing on Arduino, as well as reduce costs by removing the need of adding an extra board.

Anyway, this issue refered to the need of having any usable robot with documentation, rd2 (+ardubot or whatever is finally used) is perfectly suitable to solve this issue.

Btw, we have thought about using the new [Raspberry Pi 2](http://www.raspberrypi.org/raspberry-pi-2-on-sale/) or the [ODroid C1](http://www.hardkernel.com/main/products/prdt_info.php) as our choice of SBC, as it seems that that format of board with those holes and connectors is likely to become a standard among SBCs, and that will make easier to switch between one or other.

jgvictores, 2016-04-20:
Done at http://asrob.uc3m.es/index.php/C%C3%B3mo_montar_el_robot_RD2#Crear_SD_con_Raspbian_y_robotDevastation-robots.2Fraspi_software, duplicate of #5.

closed 2016-04-20 by jgvictores