asrob-uc3m/robotDevastation-robots #4: Battery Status

Rodriwp, 2015-02-08:
For the rd2 version with ardubot will be nice to have feedback about the state of the batteries. This version use lipo bateries for the motors and is dangerous to discharge below a certain voltage(the value depends on the number of cells)

Smallest robots probably, will have an approximated voltage measure. For the biggest robots that use ardubot, there exists an voltage regulator,amperage measure and voltage measure that cost about 10 euros.

Here is a link http://www.dx.com/es/p/zndiy-bry-apm2-5-2-apm2-6-power-module-v1-0-output-w-bec-3a-xt60-plug-black-red-342902#.VNeeT_mG-Ck

http://copter.ardupilot.com/wiki/common-measuring-battery-voltage-and-current-consumption-with-apm/?lang=es

Rodriwp, 2016-10-22:
In the software part it will be useful to encapsulate in RdRobotLib some function about robot status. Right know we are only planning to  take information from the robot about battery status, but it should support in the future more robot information (gps position, gyroscope,etc). 

Im working on it right know

Rodriwp, 2016-10-22:
I propose several states for the battery that are set using an integer. 

- NOBATERRY for robots that don't use batteries.
- A value between 1 and 100 for the percentage of battery if is discharging
-  A value between 101 and 200 for the percentage of battery if is charging

jgvictores, 2016-10-22:
I was actually going to propose implementing the YARP [IBattery](http://www.yarp.it/classyarp_1_1dev_1_1IBattery.html) interface. Here's the raw of the file:
https://github.com/robotology/yarp/blob/e0ca8e6f73c37286734d06e3afb8015529ee36b9/src/libYARP_dev/include/yarp/dev/IBattery.h
I've seen they are implementing a yarpbatterygui, so this can be a good thing.

jgvictores, 2016-10-22:
Updated link in previous comment.

David-Estevez, 2016-10-22:
+1 to the existing interface.

IMHO, I think the focus should shift from the frontend side to the robot backend. It has no sense to implement the software support for battery status if we are not currently able to measure the battery status.

Rodriwp, 2016-10-22:
+1 to the existing interface too

@David-Estevez we already have a battery mesuared system. There is one in our closet

[APM POWER MODULE](https://es.aliexpress.com/store/product/Free-Shipping-airplane-APM-Power-Module-5-3V-DC-BEC-XT60-Connectors-for-ARDUPILOT-APM-2/2218173_32711885701.html?spm=2114.04010208.3.11.mt4uPA&ws_ab_test=searchweb0_0,searchweb201602_3_10065_10068_10069_10084_10083_10017_10080_10082_10081_10060_10061_10062_10056_10055_10054_10059_10078_10079_10073_10070_421_420_10052_10053_10050_10051,searchweb201603_7&btsid=33a392cf-9d4d-4365-a3bb-e7da754c687b)

This module is at the same a time a power regulator the will solve our two battery problem and it have also voltage and current mesuare.

And the best thing, it only cost 5 $

jgvictores, 2016-10-22:
> And the best thing, it only cost 5 $ 

:joy::joy::joy:

David-Estevez, 2016-10-22:
What I meant is that the most sensible approach, in my opinion, is a bottom-up approach: hardware -> firmware -> software , not the other way around. 

Nevertheless, feel free to add support for such a magnificent piece of Chinese engineering, it will definitely be useful to have such data.

jgvictores, 2016-11-09:
@Rodriwp Could you please link to the current selected hardware?

Rodriwp, 2016-11-09:
I started some test with this device. It have six pins with the following pinout:

- 1 VCC +5
- 2 VCC +5
- 3 VCurrent
- 4 VBat
- 5  GND
- 6  GND
[Imagen Dispositivo](http://img2.banggood.com/thumb/large/upload/2014/03/SKU119430.0.jpg)
[PinOut](http://4.bp.blogspot.com/-L-IGAUheI_w/VfKFZQ2tWRI/AAAAAAAABMM/ztc3dOWLcUo/s1600/APM%2BPower%2Bmodule%2Bpinout.jpg)

Also, I have a code that can read the voltage of a battery using an analog input. As some of our robots, don't have analog input(raspberry pi), we can use an atiny85 to map de data and send via serial port.

This module is at the same time a power regulator, so we can use the same baterry for the motors and the raspberry pi.

Rodriwp, 2016-11-09:
@jgvictores  this is a link to the product [POWER MODULE 1.0](https://es.aliexpress.com/store/product/Wholesale-1pcs-2-5-APM-Flight-Controller-Power-Module-V1-0-Output-BEC-3A-Deans-T/603538_32547524353.html?spm=2114.30011108.3.3.D7G8MJ&ws_ab_test=searchweb0_0,searchweb201602_2_10092_10091_10090_10088_10089,searchweb201603_1&btsid=2feb4319-315b-4998-b77c-53a0af2943b0)

jgvictores, 2016-11-09:
@Rodriwp Great, now could you link to a URL that describes the pinout?

Rodriwp, 2016-11-09:
@jgvictores you can find the schematics of the module [here](https://upverter.com/erlerobot/a9bdc322cf64f8a1/APM25-Power-Module-v10/)

Rodriwp, 2016-11-09:
@jgvictores in a comment below you cand find the pinout. Here the link [PinOut](http://4.bp.blogspot.com/-L-IGAUheI_w/VfKFZQ2tWRI/AAAAAAAABMM/ztc3dOWLcUo/s1600/APM%2BPower%2Bmodule%2Bpinout.jpg)

Rodriwp, 2016-11-09:
Extra information about our device:
The schematics of the module [here](https://upverter.com/erlerobot/a9bdc322cf64f8a1/APM25-Power-Module-v10/)

jgvictores, 2016-11-09:
@Rodriwp As general questions related to the battery (NOT the power module):
1. What is the battery voltage range (Vmin-Vmax) we are expecting to read?
2. What is the battery current range (Imin-Imax) we are expecting to read

Rodriwp, 2016-11-23:
1. We are expecting between 3.7 and 4.1 volts per celd, and we excpected 2 and 3 cell bateries
2. We are expecting between 0 A and about 0.6 A for standar robots. More powerfull robots as RD glory, needs about 1.2A. We should support both ranges.

jgvictores, 2016-11-23:
Please focus on one specific model. Which one?

Rodriwp, 2016-11-23:
We can focus on 2 cells 0.8 max A robots

David-Estevez, 2016-11-23:
Please specify also the estimated current comsuption per device (e.g. SBC, motors (nominal and allowed peaks), camera, etc).

Rodriwp, 2016-11-23:
The RD2 have the following parts:

1. Motors: 200mA peak 300mA x 2 motors
2. Camera: 200mA
3. RaspberryPi: 800mA
4. WifiModule: 215mA

Total: 1.8 A of current expected.

jgvictores, 2016-11-23:
Okay, so that would be...
1. Sum of peak: 1.815 A
2. Sum of nominal: 1.615 A

Rodriwp, 2016-11-23:
The current arquitecture connects the motors directly to the battery, and all the other stuff through a 5V regulator.

[image]

jgvictores, 2016-11-23:
Could you please provide a link to the motors, which includes the maximum voltage?

Rodriwp, 2016-11-23:
The operating voltage is between 4.8 and 7.2

Rodriwp, 2016-11-23:
I propouse to use this solution
http://controlrobotics.rodrigomompo.com/2013/10/tutorial-battery-charge-indicator.html

jgvictores, 2016-11-23:
> The operating voltage is between 4.8 and 7.2

Seems the schematic would burn these motors.

> I propouse to use this solution

I understand you do not like your original proposal(s). Could you please upload an image of a new version of the schematic (including all components e.g. motors and RasPi) as a post on this issue?

Rodriwp, 2016-11-23:
As the current configuration of RD is deprecated, we should develop for RD ambassador #21. 
Furthermore, I think the batery status, for simplicity could take into account only voltage measure. 

For this aproach we only need to measuare the battery voltage before all the robot. So, we really don't care about the complete configuration.

jgvictores, 2016-11-23:
> As the current configuration of RD is deprecated, we should develop for RD ambassador #21.
Furthermore, I think the batery status, for simplicity could take into account only voltage measure.

The robot was never specify in this issue. Could you specify the robot in this issue?

BTW, there are no deprecated RD configurations I'm aware about. Could you please point to this fact if it exists?

> For this aproach we only need to measuare the battery voltage before all the robot. So, we really don't care about the complete configuration.

That's a design decision you can specify as the creator of this issue. Could you specify this as your objective?

Rodriwp, 2016-11-23:
RD2 use two batteries. Our current developments use only one battery. This is the reason why I said is deprecated

The objective of this issue is to measure the voltaje of the battery, as a digital output. We want to know when the battery is near to the dangerouts zone (7.6 v) and when we need to disconect the battery (7.4v)

jgvictores, 2016-11-23:
> RD2 use two batteries. Our current developments use only one battery. This is the reason why I said is deprecated

RD2 not dead! :smile:
I would definitely not DEPRECATE [our most documented robot](http://asrob.uc3m.es/index.php/C%C3%B3mo_montar_el_robot_RD2) yet.

> The objective of this issue is to measure the voltaje of the battery, as a digital output. We want to know when the battery is near to the dangerouts zone (7.6 v) and when we need to disconect the battery (7.4v)

This seems reasonable, at least to me.

David-Estevez, 2016-11-23:
> The objective of this issue is to measure the voltaje of the battery, as a digital output. We want to know when the battery is near to the dangerouts zone (7.6 v) and when we need to disconect the battery (7.4v) 

Usually when using the robot the battery voltage decreases. So 7.6V would never be a problem in normal use. And if we disconnect at 7.4V (the typical battery value)  we wont ever play.

Rodriwp, 2016-11-23:
We should check the values

We already have one module working, we can use it for testing the firmware and software implementation

jgvictores, 2016-11-23:
What do you mean by checking?

Which module do you mean?

Rodriwp, 2016-11-23:
We should adjust the alarm values to the safety lipo operation range

The circuit that is explain in http://controlrobotics.rodrigomompo.com/2013/10/tutorial-battery-charge-indicator.html, we have that circuit with us

jgvictores, 2016-11-23:
> We should adjust the alarm values to the safety lipo operation range

What do you mean by  that?

> The circuit that is explain in http://controlrobotics.rodrigomompo.com/2013/10/tutorial-battery-charge-indicator.html, we have that circuit with us

Could you please provide a schematic that includes the components (a paper version like the previous would do)?

Rodriwp, 2016-12-15:
I won't continue this idea as it lead to stalemate

David-Estevez, 2017-06-14:
I guess it is the moment to really solve this.

David-Estevez, 2017-06-14:
https://www.youtube.com/watch?v=md3B3I7Nmvw

jgvictores, 2017-06-15:
> Lashing out the action, returning the reaction
[...]
Battery is found in me
Battery
Battery

JavierIH, 2017-06-15:
:popcorn:

jgvictores, 2018-05-13:
Ongoing PR: https://github.com/robotology/yarp/pull/1583

jgvictores, 2018-06-22:
Moved to https://github.com/asrob-uc3m/robotDevastation-research/issues/11

closed 2018-06-22 by jgvictores