asrob-uc3m/robotDevastation-robots #7: RD3 and RDTurret

JuanFdelArco, 2016-04-06:
New robot design assigned to Juan F. del Arco.

David-Estevez, 2016-04-06:
@JuanFdelArco when you are more o less done with the design of the new RDTurret I can help you setup a stereo camera to have a 3D First Person View of the Turret's point of view with a phone (and old project of mine) :smile:

JuanFdelArco, 2016-04-13:
Idea #1: 
- Omnidirectional robot. 
- There are QR's on each corner.
- It can be combined with RDTurret for two players on one robot.

[image]

David-Estevez, 2016-04-13:
@JuanFdelArco Very cool robot!!

David-Estevez, 2016-04-13:
Suggestion for the wheels: 
* http://www.robotshop.com/en/58mm-plastic-omni-wheel-lego-nxt-servo.html
* http://www.robotshop.com/en/48mm-omniwheel-lego-nxt-servo.html

David-Estevez, 2016-06-01:
Looking for "robot tracks" in aliexpress: http://es.aliexpress.com/store/product/Free-shipping-smart-Car-Smart-Car-Chassis-Robot-Tracking-Sensor-Coded-Disc/1460814_32288410589.html

David-Estevez, 2016-06-02:
More ideas: use stock batteries from consumer electronics products (such as cameras, smartphones, etc) 

Example: http://es.aliexpress.com/store/product/2pcs-Battery-Charger-7-4V-1800mAh-EN-EL9-EN-EL9-ENEL9-Rechargeable-Camera-Battery-For-Nikon/1163277_1797911243.html

David-Estevez, 2016-06-14:
Even more ideas: [Cool Fighting Robots](https://www.youtube.com/watch?v=BgOG5te3jlc) via @JavierIH 

(Maybe all this info should be moved to [robotDevastation-research](https://github.com/asrob-uc3m/robotDevastation-research))

jgvictores, 2018-06-22:
This issue was moved to asrob-uc3m/robotDevastation-research#10

closed 2018-06-22 by jgvictores