asrob-uc3m/robotDevastation-robots #8: Robot (first time) configuration through access point

David-Estevez, 2016-04-06:
Suggestion for a cool improvement: When you first turn on you robot, it creates a wireless access point that you can use to configure the robot and tell which network it should connect to. Then, the robot switches off this access point and connects to the game network.

jgvictores, 2017-02-17:
Blocked by #29

jgvictores, 2017-03-29:
Unblocked as #29 closed.

The proposed interface should modify the installed versions of [these launch files](https://github.com/asrob-uc3m/robotDevastation-robots/tree/fd9f832c0a2a62b61b27aa2062626e6d0901e8bf/share/launch), which are currently hard-coded to be used from `/usr/local/share/robotDevastation-robots/contexts/launch/` in [the services](https://github.com/asrob-uc3m/robotDevastation-robots/tree/fd9f832c0a2a62b61b27aa2062626e6d0901e8bf/scripts/etc/service).

As an additional reference, we still have [this web snippet](https://github.com/asrob-uc3m/robotDevastation-robots/tree/fd9f832c0a2a62b61b27aa2062626e6d0901e8bf/web), but it lacks editing permissions on the installed files.

jgvictores, 2018-04-13:
Historic from http://asrob.uc3m.es/index.php/C%C3%B3mo_instalar_el_software_de_RD1_sobre_Raspbian_7 ([perma](http://asrob.uc3m.es/index.php?title=C%C3%B3mo_instalar_el_software_de_RD1_sobre_Raspbian_7&oldid=11274)):
```
git clone https://github.com/asrob-uc3m/robotDevastation-robots.git  # Download the robotDevastation-robots software from the repository
cd robotDevastation-robots/rd1/software; mkdir build; cd build; cmake ..  # Configure the robotDevastation-robots RD1 software
make  # Compile
sudo make install  # Install :-)

#- Extra robotName config server:
sudo apt-get install php5
sudo ln -s web/index.php /var/www
sudo ln -s web/save.php /var/www
```

jgvictores, 2019-05-07:
Setting up a Raspberry Pi as an access point in a standalone network (NAT): https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md