asrob-uc3m/robotDevastation-robots #9: Use ESP8862 as wifi module for raspberry pi 2

Rodriwp, 2016-04-07:
In order to decrease latency, test if this modules are better than the current ones

David-Estevez, 2016-04-07:
ESP8266 modules aren't USB WiFi dongles, but bridges between USARTs and TCP/UDP sockets on the PC side. Therefore, I doubt that they can be used to stream images from the raspberry pi (and if they can, it's probably through a even slower connection).

To reduce latency (if that is the issue), two approaches can be followed:

1. Upgrade cheap WiFi dongle with a higher quality (Bandwidth) one. I can lend you one ASUS N11 dongle I have at home for testing.
2. Try to view a mjpeg stream directly using mjpg-streamer on the raspberry pi. If using only mjpg-streamer the video doesn't have any latency, this latency might be introduced by the image processing done on the PC (not related to WiFi hardware).

jgvictores, 2016-04-07:
Option 2 involves invoking mjpg_streamer on the raspi, as in:
`mjpg_streamer -i "input_uvc.so -f 15 -d /dev/video0 -y" -o "output_http.so -p 8080"`
It's good because we'd drop some wrapping processes involved in using opencv and yarp on the raspi side. We could then hook the stream on the robotDevastation (PC client) side as we did in https://github.com/robotology/yarp/issues/254 (with a Darwin-op robot using mjpg_streamer: https://sourceforge.net/projects/mjpg-streamer/).

David-Estevez, 2016-04-07:
Another possible related software improvement could be print the received images _before_ further processing (to have a higher update rate) and process only a few frames (use tracking to interpolate enemies between frames).

jgvictores, 2016-04-08:
Ok to change the topic to **Decrease video transmission latency**?

Rodriwp, 2016-04-10:
RaspiMjpeg seems to solve the problem. 
https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=61771
Next sesion I will test it

David-Estevez, 2016-04-23:
Some new points:

1. Last wednesday we tested the wifi connection and it was ok between two PCs on the same LAN, so the problem must be related to the robot. Next thing to try is changing the wifi dongle by a faster one, to check if high latency is related to wifi dongle bandwidth.
2. Related to use the ESP8266 as wifi dongle, I have found the following resources:
    * https://hackaday.io/project/8678-rpi-wifi
    * https://github.com/al177/esp8089

    It is a difficult issue, and I'm not sure that this solution is faster that the current chinese wifi dongle.

Rodriwp, 2016-11-09:
The next test we need to perform is with RD2 Black and @David-Estevez expensive wifi module. So, we can discard that our chinese hardward is the problem

Rodriwp, 2016-11-23:
@David-Estevez  and @jgvictores Did you realize if RD ambassador decrease transmission latency ?

jgvictores, 2016-11-23:
Yesteday at https://github.com/asrob-uc3m/actas/issues/36 latency was surprisingly low.

It would be great if you could stay focused on #4 for the time being.

jgvictores, 2018-05-08:
Just for the record, I've read one of the greatest improvements on `Raspberry 3 B+` is on comms, note the 2.4GHz and 5GHz IEEE 802.11.b/g/n/ac wireless LAN spec: https://www.raspberrypi.org/products/raspberry-pi-3-model-b-plus/

jgvictores, 2018-06-22:
This issue was moved to asrob-uc3m/robotDevastation-research#13

closed 2018-06-22 by jgvictores