asrob-uc3m/robotDevastation-robots #13: Use ARM Cortex M

Rodriwp, 2016-05-09:
I buy for 2$ an ARM Cortex M3. As this hardware seems to be more powerfull, let's test if we can do ip video with it. Probably a bundle arm cortex m3 + esp8266 could be the core of the new low cost robots.

If M3 is not powerfull enoght we can test M4 and M7. With this we can create the robot devastation board.

David-Estevez, 2016-05-09:
This comment has two parts: one in which I warn you about why this is difficult and another in which I provide you some resources just in case you still want to try to do this.

First of all, if you want to have a achieve transmission rate, you won't be using Arduino-like libraries to program the ARM chip, but manipulating raw registers and memory. You will have to find out what is the protocol for image transmission in the camera module, what is the image format, and what is the compression algorithm used (if it has any). Then,  you will have to implement a socket-based streamer to transmit the image data. You may have or not to encode the data in each packet, but still you will have to implement all the streamer code (take a look of mjpg-streamer code, for instance) in the ARM chip. Take into account also debugging of the code, which will be also difficult as it involves several pieces of hardware and image processing.

If after all of this you still want to build this device, here you have some resources that might be useful for you:

* This [post in hackadaty](https://hackaday.com/tag/wifi-camera/) shows an existing implementation of a DIY WiFi camera done with the M3 in the ESP8266 board.
* This comment in other related issue: https://github.com/asrob-uc3m/robotDevastation-robots/issues/9#issuecomment-213746159 gives you some clues about the WiFi transmission part.

David-Estevez, 2016-05-09:
FYI: http://es.aliexpress.com/store/product/WIFI-1280-x-960P-1-3MP-Mini-IP-Camera-Indoor-Black-Security-Camera-Night-Vision-ONVIF/415422_32573773191.html

Rodriwp, 2016-05-09:
Probably It will be nice to use 0V7670 camera. It cost around 3'50$ and is a well know camera with a lot of information. A full robot devastation pcb with 4 motor ports and a battery could be achive between 10$ and 15$ if this works.

Rodriwp, 2016-11-23:
As raspberry pi zero cost only 5$ and the purpose of this issue is deacrease the cost, right know it not make sense this aproach.

closed 2022-06-04 by PeterBowman