asrob-uc3m/robotDevastation-robots #26: Improve launching robots: decide on which to focus.

jgvictores, 2016-12-11:


David-Estevez, 2016-12-11:
The original roadmap proposed working with "wired" robots (i.e. turrets) because they are (in principle) easier to setup and easier to work with when developing the game.

So I vote for working on turret-like robots first, and them move on remote robots. 

It is outside the scope of this repo, but we also have to have in mind (milestone 5)[https://github.com/asrob-uc3m/robotDevastation/milestone/5] from RobotDevastation main repo.

jgvictores, 2017-02-15:
Software-wise:
- Locally connected turrets are supported, see [Launch Robot Devastation - With Fake Motors And With PC Webcam](https://github.com/asrob-uc3m/robotDevastation-user-manual/blob/a91657b481702fd1a795322e68d017efca3b3665/en/launch-robot-devastation---LaserTowerOfDeath-with-webcam.md). ToDo:
  - [ ] https://github.com/asrob-uc3m/robotDevastation/issues/91
- Remotely connected turrets have multiple issues. To name some:
  - [ ] #29 
  - [ ] #9

jgvictores, 2017-05-24:
More related: #21 #34

jgvictores, 2017-12-01:
Will focus on simulated robots for now, see #36

closed 2017-12-01 by jgvictores