asrob-uc3m/robotDevastation-robots #29: Replace yarprun by list of services parsed from file

jgvictores, 2017-02-15:
Replace yarprun by list of services (motors, camera...) parsed from file.

The file could be modified by a service such as the proposed in #8.

jgvictores, 2017-02-15:
Current scripts can contain a `yarp detect`.

David-Estevez, 2017-02-16:
Just in case... To setup the autostart of a wiki in my server I used [systemd](https://wiki.archlinux.org/index.php/Systemd) and the command `systemctl` successfully.

jgvictores, 2017-03-08:
Progress being done on black RD2 "robotera", in branch `issue-29-replace-yarprun`, around 9c27066f39cc93f71c00fa08f3aa1b5865407d7.
Progress is slow due to compilation on Raspi1 (bump #11).
1. Compiled [yarp 2.3.68+181-20170203.11+gite122128](https://github.com/roboticslab-uc3m/installation-guides/blob/4a9cf4ca9d3a50b5237d8248faaeaaf1de0ed635/install_yarp.md#install-yarp-2368-debian-6010) to avoid compilation errors.
2. Am recompiling without `SKIP_ACE`, because shared libraries were not being loaded.

jgvictores, 2017-03-15:
Help: https://isotope11.com/blog/manage-your-services-with-daemontools

jgvictores, 2017-03-22:
Replaced `yarprun` service by two services (`launchCamera` and `launchRobot`) at c65f8bc3ce9e65db5bd3556b99c81cf495dfc816, hard-coded for `RaspiOnePwmMotorController` used on black RD2 (a.k.a. `rd2old` and `robotera` :smile:).
Next steps:
- [ ] Document process (or automate!) of copying services from `robotDevastation-robots/scripts/etc/service` to OS '/etc/service`.
- [ ] Parse config from file as proposed by this issue.

jgvictores, 2017-03-29:
Done at fd9f832c0a2a62b61b27aa2062626e6d0901e8bf.

closed 2017-03-29 by jgvictores