asrob-uc3m/robotDevastation-robots #32: Smaller Raspberry image size

David-Estevez, 2017-03-29:
Since we are not using the whole Raspian (think about the graphical interface, for instance) we can decrease the size of the image for the release.

Refs:
* http://www.cnx-software.com/2012/07/31/84-mb-minimal-raspbian-armhf-image-for-raspberry-pi/
* https://github.com/andrius/build-raspbian-image
* http://www.cnx-software.com/2012/07/31/84-mb-minimal-raspbian-armhf-image-for-raspberry-pi/

jgvictores, 2017-12-02:
Haven't seen refs, but this could be done via `fsarchiver` as done [elsewhere](https://github.com/roboticslab-uc3m/yarp-devices/issues/43).

jgvictores, 2019-06-25:
Just to clarify:
- The `dd` mechanism used at time of writing creates larger images than `fsarchiver` which is being studied at time of writing
- The original purpose of this issue is effectively reducing the OS disk usage size, e.g. by removing GUI components, which would directly result in a smaller image via `fsarchiver`

jgvictores, 2020-05-17:
Update: `fsarchiver` was successfully used at https://github.com/asrob-uc3m/rd-ambassador/issues/1#issuecomment-503223715