asrob-uc3m/robotDevastation-robots #35: Cool mobile robot

David-Estevez, 2017-05-31:
Rd Ambassador was ok, but it has several flaws that are hard to solve, the main one being that there are too many things to put on the mobile platform, and not enough space (camera, QR, Raspberry Pi...).

So I have started the development of a cool mobile robot to be the documented basic robot to be used in Robot Devastation.

The basic features it will have are:

* Instead of Raspberry Pi, it will use an Orange Pi Lite. The advantages of these board are the low cost, small size, and integrated WiFi card. We are also researching the possibility of using a camera connected to the camera bus of the board (does not require USB).
* Camera (either USB webcam or camera connected to camera bus).
* A shield for the Orange Pi with the following elements:
    * H-bridge or similar motor controller for 2 DC motors
    * Battery status (related to #4 ) feedback
    * Regulator (which might me placed outside the board if we find that it is a better solution).
    * LEDs for status feedback
    * Buttons for user input
* It will use consumer camera batteries (or similar), that are easy to source as replacements, cheap and have their own safe and easy-to-use charger.

jgvictores, 2017-05-31:
- https://forum.armbian.com/index.php?/topic/3054-quality-of-images-from-orange-pi-pc-camera-module/
- http://192.241.128.247/2016/04/05/orange-pi-camera-with-v4l2loopback-vidcopy/

jgvictores, 2017-05-31:
- http://192.241.128.247/2016/04/05/orange-pi-camera-with-v4l2loopback-vidcopy/ -> [orangePiCamera.pdf](https://github.com/asrob-uc3m/robotDevastation-robots/files/1042437/orangePiCamera.pdf)

jgvictores, 2017-05-31:
```
./vidcopy -w 800 -h 600 -r 20 -i /dev/video0 -o /dev/video1 -f UYVY
Required width: 800
Required height: 600
Required FPS: 20
input device: /dev/video0
output device: /dev/video1
Driver Caps:
  Driver: "sunxi-vfe"
  Card: "sunxi-vfe"
  Bus: "sunxi_vfe sunxi_vfe.0"
  Version: 1.0
  Capabilities: 05000001
  FMT : CE Desc
--------------------
  422P:    planar YUV 422
  YU12:    planar YUV 420
  YV12:    planar YVU 420
  NV16:    planar YUV 422 UV combined
  NV12:    planar YUV 420 UV combined
  NV61:    planar YUV 422 VU combined
  NV21:    planar YUV 420 VU combined
  HM12:    MB YUV420
  YUYV:    YUV422 YUYV
  YVYU:    YUV422 YVYU
  UYVY:    YUV422 UYVY
  VYUY:    YUV422 VYUY
  BA81:    RAW Bayer BGGR 8bit
  GBRG:    RAW Bayer GBRG 8bit
  GRBG:    RAW Bayer GRBG 8bit
  GRBG:    RAW Bayer RGGB 8bit
  BG10:    RAW Bayer BGGR 10bit
  GB10:    RAW Bayer GBRG 10bit
  BA10:    RAW Bayer GRBG 10bit
  BA10:    RAW Bayer RGGB 10bit
  BG12:    RAW Bayer BGGR 12bit
  GB12:    RAW Bayer GBRG 12bit
  BA12:    RAW Bayer GRBG 12bit
  BA12:    RAW Bayer RGGB 12bit
Selected Camera Mode:
  Width: 640
  Height: 480
  PixFmt: UYVY
```

David-Estevez, 2017-06-04:
After reading the post a bit more in detail, it seems that there is a parameter, `VIDIOC_S_INPUT` that has to be configured for this camera to work, but most v4l2-based programs leave unconfigured by default.

I adapted his function `v4l2_set_input` to work with `mjpeg-streamer`'s `input_uvc` plugin, and called it before the initialization of the camera. This removes the error from the `mjpeg-streamer`'s output and allows streaming of the camera without `v4l2-loopback`. The adapted function looks like this:
```cpp
int v4l2_set_input(struct vdIn *dev) {
	/*
	 * define video input
	 * vfe_v4l2 the driver is forced to input = -1
	 * set as the input = 0, works fine.
	 */
	struct v4l2_input input;
	int count = 0;
        memset(&input, 0, sizeof(struct v4l2_input));
	input.index = count;
	while(!xioctl(dev->fd, VIDIOC_ENUMINPUT, &input)) {
		input.index = ++count;
	}
	count -= 1;

        if(count <= -1)
        {
            printf(stderr, "Error with counter (<=-1)\n");
           return -1;
        }

	if(xioctl(dev->fd, VIDIOC_S_INPUT, &count) == -1) {
		print_debug("Error selecting input %d", count);
		return 1;
	}
	return 0;
}
```

Now I'm a bit worried about latency with this camera, since it is streaming with UYUV enconding, which has a lower framerate and larger latency than streaming directly in mjpeg. 

But, since streaming works with both webcams and the OPi SH-2035 Cam, the next important think to check with this board is GPIO control via WiringOP to control h-bridges and other hardware required to control the robot.

David-Estevez, 2017-06-04:
It seems that it is indeed possible to harness the power of HW H264 encoding to [stream video with ffmpeg]( https://forum.armbian.com/index.php?/topic/1842-ffmpeg-with-cedrus-h264-hw-encoder-h3-cmos-camera/#entry14157). It might be interesting to test that as it could potentially solve most of our problems with latency. 

Other choice with support for HW H264 enconding and similar characteristics as the OPi it the [RPi Zero W](https://www.raspberrypi.org/products/pi-zero-w/). You can even [emulate a network connection through OTG](https://gist.github.com/gbaman/975e2db164b3ca2b51ae11e45e8fd40a) to have a ssh connection similar to the serial tty present by default on the OPi.  Still, the [camera module for the RPi](https://www.raspberrypi.org/documentation/hardware/camera/README.md) is a bit expensive (~25$) compared with the OPi camera.

David-Estevez, 2017-06-06:
GPIOs work as expected using [WiringOP](https://diyprojects.io/orange-pi-armbian-install-wiringop-library-wiringpi-equivalent/). The original code for WiringPI  works almost without changes, although there are some link errors that require to explicitly specify the library to link to. (That code is still on the robot, as it is a hack).

jgvictores, 2017-06-12:
Back on webcam topic, attaching orange-pi-camera (left) vs cheap webcam (right) screenshot.

[image]
.

David-Estevez, 2017-06-13:
Note that both images were taken at the same distance from the target. I would say that the first one (OPi camera) has better FOV but lower quality / focus whereas the other one (cheap webcam) has a pretty bad FOV but everything is in focus.

David-Estevez, 2017-06-15:
I'm currently working on #4 to provide the new cool robot the ability to know how much battery does it have left.

In addition, I have created the [Cool new robot project](https://github.com/asrob-uc3m/robotDevastation-robots/projects/1) to organize myself (and however is working on this with me at the moment) given all the features required to close this issue.

jgvictores, 2018-06-22:
This issue was moved to asrob-uc3m/robotDevastation-research#12

closed 2018-06-22 by jgvictores