asrob-uc3m/robotDevastation-robots #46: Define state, clean, and decide fate of this repo

jgvictores, 2018-11-26:
Stemming from https://github.com/asrob-uc3m/rd2/issues/1#issuecomment-441367317, this is a tracking issue to define the state, clean, and decide fate of this repo.

Here is a list of what we have as of current `develop` 6bfe26b456a0f03cb34ebccf6b6c2b9eae662844:
1. firmware/extern/MT7601: will be exported as a result of #45
2. A list pointing to robot repositories
3. Some generated QRs
4. An idea of a mini-web service, see #8
5. Shared ini files

Spoken with @PeterBowman, if we wanted to destroy this repo:
1. firmware/extern/MT7601: will be exported as a result of #45
2. A list pointing to robot repositories: could be moved to robotDevastation wiki or robotDevastation-user-manual
3. Some generated QRs: ?
4. An idea of a mini-web service, see #8: ?
5. Shared ini files: This in fact involves a different decision on how/where we want these files:
   1. Centralized (robotDevastation-configuration-files)
   2. robot-specific, because it is very robot specific except for port name (however: we could recover pre- #29 `robotName` file)

PeterBowman, 2019-04-18:
Doxygen stuff removed at https://github.com/asrob-uc3m/robotDevastation-robots/commit/66690e3d6dd894fe5ef8ab1198526c35f04bfa8f.

jgvictores, 2020-05-17:
Idea: repo name change to `robotDevastation-robot-configuration-files` or simply `robotDevastation-configuration-files`.

The existing robot list could be moved to: any manual (user/developer), the main page, or even to a separate page like https://robots.ieee.org/ :grin:

jgvictores, 2020-05-17:
> Idea: repo name change to `robotDevastation-robot-configuration-files` or simply `robotDevastation-configuration-files`.

Note that this was already proposed above :see_no_evil: