asrob-uc3m/robotDevastation-robots #47: Design robot that combines best of RD1 and RD2

jgvictores, 2019-03-27:
Design robot that combines:
- Best of RD1:
    - Standard battery holder for standard batteries for motor
    - Use any mobile phone battery for raspi
    - FreeCAD design
- Best of RD2:
    - Webcam support
    - QR support
    - 3rd support for stability (a metal marble on RD2)

Leytha, 2019-06-18:
Camera Holder similar to RD2 one but as a FreeCAD design: 
[CameraHolder.fcstd.zip](https://github.com/asrob-uc3m/robotDevastation-robots/files/3303002/CameraHolder.fcstd.zip)

jgvictores, 2020-05-17:
Provided https://github.com/asrob-uc3m/robotDevastation-robots/issues/46#issuecomment-629783236, this issue and similar could be moved to https://github.com/asrob-uc3m/robotDevastation-playground