asrob-uc3m/robotDevastation-robots #48: WiringPi library about to be deprecated

David-Estevez, 2019-09-19:
FYI: https://hackaday.com/2019/09/18/wiringpi-library-to-be-deprecated/

jgvictores, 2019-09-19:
@David-Estevez Thanks for the info!

PeterBowman, 2019-09-19:
Pity we never forked git://git.drogon.net/wiringPi into the asrob-uc3m org, I would do that as soon as the server is up again.

PeterBowman, 2019-09-20:
@jgvictores found a local clone of drogon.net, now it lives in our org at https://github.com/asrob-uc3m/wiringPi.

Next, I'm going to unbreak Travis jobs by fetching these sources instead.

PeterBowman, 2019-09-20:
> Next, I'm going to unbreak Travis jobs by fetching these sources instead.

Done at https://github.com/asrob-uc3m/yarp-devices/commit/44921d4ad9fe707d36ee03693d9293b0f17e9332.

closed 2019-09-20 by PeterBowman

jgvictores, 2020-05-18:
Updated in our install doc at https://github.com/asrob-uc3m/yarp-devices/commit/7b3411f594dbaa288470811ffdf17021135c0785